FabioBatSilva/ArduinoFake #1: Compile error in example wiring-blink

No3x, 2017-06-17:
I want to play around with your testing framework but can't compile the example `wiring-blink`.
I get this error for each occurence of `std::strstr`:
```cpp
.piolibdeps\ArduinoFake_ID1689\src/ArduinoFake.h:33:14: error: 'strstr' is not a member of 'std'
if (std::strstr(typeid(*var).name(), #mock)) { \
^
.piolibdeps\ArduinoFake_ID1689\src/ArduinoFake.h:79:13: note: in expansion of macro 'ArduinoFakeReturnInstaceOf'
ArduinoFakeReturnInstaceOf(p, Serial)
^~~~~~~~~~~~~~~~~~~~~~~~~~
```
I'm on windows with mingw-w64: gcc-7.1.0
platform information: https://pastebin.com/ZCCbF3E9

I did not find any information on this. Maybe you know the reason?

FabioBatSilva, 2017-06-17:
Hi @No3x,

I'm guessing `strstr` is not part of the `std` in mingw-w64: gcc-7.1.0
It might be missing a `#include <cstring>` here [ArduinoFake](https://github.com/FabioBatSilva/ArduinoFake/blob/master/src/ArduinoFake.h#L13).

Can you give that a try ?

No3x, 2017-06-17:
I already tried this lines in all variants:
```
#include <iostream>
#include <cstring>
using namespace std;
```
without any success. 

EDIT: gosh.. I have edited the 
`ArduinoFake\src instead` the
`ArduinoFake\examples\wiring-blink\.piolibdeps\ArduinoFake_ID1689\src`
all the time..

FabioBatSilva, 2017-06-17:
It might not be using the same version..
What do you get when running the tests in the root directory ?

```
cd path-to/ArduinoFake
pio test

```

No3x, 2017-06-17:
pio test in the root fails like before.
But with 
```cpp
#include <iostream>
#include <cstring>
```
added it does not.

FabioBatSilva, 2017-06-17:
Cool..

Can you send a PR ?

No3x, 2017-06-17:
Of course

No3x, 2017-06-17:
```bash
pio test
==================== [test::*] Building... (1/2) ======================
Please wait...
C:/Program Files (x86)/mingw-w64/i686-7.1.0-posix-dwarf-rt_v5-rev0/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.1.0/../../../../i686-w64-mingw32/bin/as.exe: .pioenvs\native\test\main.o: too many sections (36801)
C:\Users\user\AppData\Local\Temp\ccobVkGB.s: Assembler messages:
C:\Users\user\AppData\Local\Temp\ccobVkGB.s: Fatal error: can't write 117 bytes to section .text of .pioenvs\native\test\main.o because: 'File too big'
C:/Program Files (x86)/mingw-w64/i686-7.1.0-posix-dwarf-rt_v5-rev0/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.1.0/../../../../i686-w64-mingw32/bin/as.exe: .pioenvs\native\test\main.o: too many sections (36801)
C:\Users\user\AppData\Local\Temp\ccobVkGB.s: Fatal error: can't close .pioenvs\native\test\main.o: File too big
*** [.pioenvs\native\test\main.o] Error 1
```

Could fix this by setting build flags to `build_flags = -std=gnu++11 -O3 -Wa,-mbig-obj`
https://stackoverflow.com/questions/16596876/object-file-has-too-many-sections
https://forum.qt.io/topic/59978/error-file-too-big/6

I really hate this C++ compiler jungle.

Sparow199, 2017-07-04:
Hello, i have added `#include <cstring>` and it works finally.

No3x, 2017-07-06:
@Sparow199 thanks for confirming this. There is still a pending PR that should fix this. I guess you don't compile on Windows/MinGW - because this introduces a new issue described in my last comment.

closed 2022-04-10 by FabioBatSilva