FabioBatSilva/ArduinoFake #2: Test interaction with mock in other class

No3x, 2017-06-17:
Hi,
I'm still reading the code to understand how it works. It's really hard stuff since I have never used templates and C++11 features before (because of missing compiler support). 
As I understand your framework relies on the include of ArduinoFake.h instead of Arduino.h while unittesting.
```cpp
#ifdef UNIT_TEST
    #include "ArduinoFake.h"
#else
    #include "Arduino.h"
#endif
```
I have a service that get a derivated instance of the `Client` injected (WiFiClientSecureRedirect in my case). I want to test the interaction with the client and fake returns of the client.
There are some points I'm not sure about:

1. - I think there is no fake for the Client yet. This would require add it analogous to `SerialFake`
2. - I don't understand how the type is swapped to the fake type (where? how?)
3. - Is it possible to test another file than the main.cpp? 

To 2. - My initial thought was you put a different e.g. Serial.h into your lib and include this one instead of the real Serial.h to put a class Serial into it and implementing the fake there. (But thats not correct obvously, since you have a File "SerialFake.h/cpp" and the class is called "SerialFake").

This is my scaffolding:
ServiceA.h
```cpp
#pragma once

#ifdef UNIT_TEST
    #include "ArduinoFake.h"
#else
    #include "Arduino.h"
#endif

#include <Client.h>

class ServiceA
{
  public:
    ServiceA (Client &client);
    String request(const char*  url);

  private:
    Client* _client;
};
```
ServiceA.cpp
```cpp
#include "ServiceA.h"

ServiceA::ServiceA(Client &client)
: _client(&client)
{}

ServiceA::request(const char *url) {
  //TODO: meaningful implementation.
  // Currently not implemented because testing/mock is the focus
  // Can I inject a client mock and verify interaction with connect?
  client->connect(url, 80);
  return url;
}
```
test_servicea.cpp
```cpp
#include "ServiceA.h"
#include <ArduinoFake.h>
#include <unity.h>

using namespace fakeit;

ServiceA* service;

void test_connect(void)
{
    When(Method(ArduinoFake(), connect)).Return();

    service->request("google.com");

    Verify(Method(ArduinoFake(), connect)).Once();
}

void setUp(void)
{
    Client clientMock = getClientMock//?
    service = new ServiceA(clientMock);
    ArduinoFakeReset();
}

int main(int argc, char **argv)
{
    UNITY_BEGIN();

    RUN_TEST(test_connect);

    UNITY_END();

    return 0;
}
```
I think Client.h is not accessible during unit tests because of missing framework/platform includes.

Can you please share your thoughts about this usecase? Is it possible this way? Maybe you can elaborate on the function of the framework.

FabioBatSilva, 2017-06-17:
> I'm still reading the code to understand how it works. It's really hard stuff since I have never used templates and C++11 features before (because of missing compiler support). 
> As I understand your framework relies on the include of ArduinoFake.h instead of Arduino.h while unittesting.

Yes,
But i'm thinking it might not be necessary.
if we have `src/Arduino.h` that includes `src/ArduinoFake.h` the change between native testing and Arduino would work seamless as long as there is no other `Arduino.h` included..  

> 1. - I think there is no fake for the `Client` yet. This would require add it analogous to `SerialFake`

Yes, it would required adding a `Client`/`ClientFake` and injecting a instance of that mock into your code..

> 2. - I don't understand how the type is swapped to the fake type (where? how?)

This lib uses the headers from Arduino but it has a different implementation.
So when your call something like `digitalRead()` your are calling a [fake implementation 
](https://github.com/FabioBatSilva/ArduinoFake/blob/master/src/FunctionFake.cpp#L14) that invokes a [mock instance](https://github.com/FabioBatSilva/ArduinoFake/blob/master/src/FunctionFake.h#L15) generated by [FakeIt](https://github.com/eranpeer/FakeIt)

> 3. - Is it possible to test another file than the main.cpp?

Yes, you just need to include that file/header in your tests

> To 2. - My initial thought was you put a different e.g. Serial.h into your lib and include this one instead of the real Serial.h to put a class Serial into it and implementing the fake there. (But [...]

Yes, i think we are on the right track ..

Notice that `Serial` is a global/extern [variable](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBAPI.h#L152) not a class.. 
When `USBAPI` exposes a instance of [Serial_](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBAPI.h#L152) it is exposing whatever implementation it has.. in this case a [SerialFake](https://github.com/FabioBatSilva/ArduinoFake/blob/master/src/SerialFake.cpp)

For `Client` you probably don't need/want a single instance,
I'm not entirely sure how that is going to play.. but we probably need a way to expose mocks/instances of Arduino classes..

`ArduinoFake(Serial)` is going to return a `fakeit::Mock<SerialFake>`
But for `ArduinoFake(Client)` it would have to return a `fakeit::Mock<Client>`

Then :
```cpp
When(Method(ArduinoFake(Client), connect)).Return();

Client client = ArduinoFake(Client).get();

ServiceA service(&client);

service.request("google.com");

Verify(Method(ArduinoFake(Client), connect)).Once();
```

No3x, 2017-06-24:
I have added a new example as scaffold for the described tests in my fork. Unfortunately I get the following linker errors:
```
.pioenvs\native\src\ArduinoFake.o:ArduinoFake.cpp:(.text.startup+0x53): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
.pioenvs\native\src\ClientFake.o:ClientFake.cpp:(.text.startup+0x49): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
.pioenvs\native\src\FunctionFake.o:FunctionFake.cpp:(.text.startup+0x49): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
.pioenvs\native\src\HardwareSerialFake.o:HardwareSerialFake.cpp:(.text.startup+0x49): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
.pioenvs\native\src\IPAddressFake.o:IPAddressFake.cpp:(.text.startup+0x49): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
.pioenvs\native\src\PrintFake.o:PrintFake.cpp:(.text.startup+0x49): more undefined references to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)' follow
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\native\program.exe] Error 1
```
Maybe you can spot the error?

FabioBatSilva, 2017-07-08:
Please check out https://github.com/FabioBatSilva/ArduinoFake/tree/master/examples/mock-injection

No3x, 2017-07-08:
Thanks @FabioBatSilva for working on this. I get a lot of compiler errors when running pio test in the root of the project.

```gnu
=============================================================================================================================== [test::*] Building... (1/2) ===============================================================================================================================
Please wait...
In file included from src\arduino\IPAddress.cpp:20:0:
src\arduino\Arduino.h:122:9: error: 'uint8_t' does not name a type; did you mean 'wint_t'?
typedef uint8_t byte;
^~~~~~~
wint_t
src\arduino\Arduino.h:129:14: error: variable or field 'pinMode' declared void
void pinMode(uint8_t, uint8_t);
^~~~~~~
src\arduino\Arduino.h:129:14: error: 'uint8_t' was not declared in this scope
src\arduino\Arduino.h:129:14: note: suggested alternative: 'wint_t'
void pinMode(uint8_t, uint8_t);
^~~~~~~
wint_t
src\arduino\Arduino.h:129:23: error: 'uint8_t' was not declared in this scope
void pinMode(uint8_t, uint8_t);
^~~~~~~
src\arduino\Arduino.h:129:23: note: suggested alternative: 'wint_t'
void pinMode(uint8_t, uint8_t);
^~~~~~~
wint_t
src\arduino\Arduino.h:130:19: error: variable or field 'digitalWrite' declared void
void digitalWrite(uint8_t, uint8_t);
^~~~~~~
src\arduino\Arduino.h:130:19: error: 'uint8_t' was not declared in this scope
src\arduino\Arduino.h:130:19: note: suggested alternative: 'wint_t'
void digitalWrite(uint8_t, uint8_t);
^~~~~~~
wint_t
src\arduino\Arduino.h:130:28: error: 'uint8_t' was not declared in this scope
void digitalWrite(uint8_t, uint8_t);
^~~~~~~
src\arduino\Arduino.h:130:28: note: suggested alternative: 'wint_t'
void digitalWrite(uint8_t, uint8_t);
^~~~~~~
wint_t
src\arduino\Arduino.h:131:17: error: 'uint8_t' was not declared in this scope
int digitalRead(uint8_t);
^~~~~~~
```
But I think my compiler just doesn't include std libs again.

No3x, 2017-07-08:
Adding `#include <stdint.h>` to the Arduino.h
And appending `-O3 -Wa,-mbig-obj` to the build flags in the platform.ini. (As described in the other issue) 
fixed it.
Sorry I'm not that familiar with cpp so I have no glue if this is a problem in general or mingw specific.

No3x, 2017-07-09:
I just moved my arduino dev to a ubuntu vm because there were too many problems with the compiler. -  `pio test` in the root works out of the box now! (never did before for me) 👍
 

But I do have 2 problems with the mock-injection example:

## 1. the dependency in the platformio.ini is not clonable
```
Permission denied (publickey).
fatal: Could not read from remote repository.
```
Detailed version:
```
========================= [test::*] Building... (1/2) =========================
Please wait...
LibraryManager: Installing ArduinoFake
git version 2.7.4
Cloning into '/home/no3x/Desktop/ArduinoFake/examples/mock-injection/.piolibdeps/_tmp_installing-rB2NeB-package'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Error: Traceback (most recent call last):
  File "main.py", line 82, in main.main (/tmp/tmpo9SPS0/main.c:2472)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "cmd_test.py", line 120, in cmd_test.cli (/tmp/tmpo9SPS0/cmd_test.c:3941)
  File "cmd_test.py", line 178, in cmd_test.cli (/tmp/tmpo9SPS0/cmd_test.c:3851)
  File "cmd_test.py", line 337, in cmd_test.NativeTestProcessor.process (/tmp/tmpo9SPS0/cmd_test.c:7707)
  File "cmd_test.py", line 257, in cmd_test.TestProcessorBase.build_or_upload (/tmp/tmpo9SPS0/cmd_test.c:6055)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 111, in cli
    result = (envname, ep.process())
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 189, in process
    result = self._run()
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 282, in _run
    ], self.verbose)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/commands/run.py", line 303, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/commands/lib.py", line 97, in lib_install
    lm.install(library, silent=silent, interactive=interactive)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/managers/lib.py", line 252, in install
    trigger_event)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 627, in install
    name, url, requirements, track=True)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 369, in _install_from_url
    assert vcs.export()
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/vcsclient.py", line 149, in export
    assert self.run_cmd(args)
  File "/home/no3x/.local/lib/python2.7/site-packages/platformio/vcsclient.py", line 96, in run_cmd
    return check_call(args, **kwargs) == 0
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
CalledProcessError: Command '['git', 'clone', '--recursive', '--depth', '1', 'git@github.com:FabioBatSilva/ArduinoFake.git', '/home/no3x/Desktop/ArduinoFake/examples/mock-injection/.piolibdeps/_tmp_installing-rB2NeB-package']' returned non-zero exit status 128
```
Changing this to `https://github.com/FabioBatSilva/ArduinoFake.git` solves it. It might work by generating an RSA key on github and register it on the ubuntu somehow but I think with https is is much more simple and elegant.

## 2. running `pio test` fails
Running `pio test` in the project root runs fine.
Running `pio test` in the example fails with an `fakeit::SequenceVerificationException`:
```
======================================= [test::*] Building... (1/2) =======================================
Please wait...

======================================= [test::*] Testing... (2/2) =======================================
terminate called after throwing an instance of 'fakeit::SequenceVerificationException'
what():  test/test_my_service.cpp:36: Verification error
Expected pattern: ArduinoFake(Client).println(STATUS)
Expected matches: exactly 1
Actual matches  : 0
Actual sequence : total of 11 actual invocations:
ArduinoFake(Client).connect(myserver.com, 80)
ArduinoFake(Client).println(STATUS)
ArduinoFake(Client).println()
ArduinoFake(Client).available()
ArduinoFake(Client).read()
...

============================================= [TEST SUMMARY] =============================================
test:*/env:native	[FAILED]
```
The same behavior when creating a whole new project and really using ArduinoFake as lib for tests.

closed 2022-04-10 by FabioBatSilva