FabioBatSilva/ArduinoFake #6: Using ArduinoFake on platformio with native platform but arduino framework

dolfandringa, 2020-02-18:
I am using (I guess as most people) ArduinoFake for unittesting arduino (avr) code on my "native" platform (gcc on linux X64). I am using it in combination with platformio and its test runner. 
There are a number of libraries I am using that should work with any arduino platform (like ArduinoJson and others). What I am not sure about is how to combine them with platformio, the native platform and ArduinoFake. In the library.json you specify `"frameworks": "arduino"` and `"platforms": "native"`. But there is no arduino framework defined in the native platform. And also the arduino framework requires a board, which there aren't for the native platform. So am I missing something here?

dolfandringa, 2020-02-18:
It seems like setting `lib_compat_mode = off` makes platformio not check if libraries are compatible with the current platform/framework. So the standard arduino libraries then will get included just fine. What I personally did is define a macro with `-DNATIVE` and then make a custom lib `MockLib.h` which itself includes `ArduinoFake.h` and I can add additional fake stuff in there as required. I am still having some issues, but this got me a lot further. Is this how this is supposed to be done?

FabioBatSilva, 2020-02-18:
I think the `lib_compat_mode` issue is related to https://github.com/FabioBatSilva/ArduinoFake/pull/5

What version of `ArduinoFake` are you using ?

dolfandringa, 2020-02-19:
> What version of `ArduinoFake` are you using ?

0.2.0

dolfandringa, 2020-02-19:
Manually setting lib_compat_mode=off in my platformio.ini works, but it doesn't happen automatically.

FabioBatSilva, 2020-02-19:
I think 0.2.0 didn't include the `libCompatMode=off`

I've tagged 0.2.1 yesterday.
Can you try bumping the version or perhaps point your dependency to git / master and test it out ?

dolfandringa, 2020-02-20:
I both installed the latest version, and then removed it and installed the git version, and with both, I need to manually specify the `lib_compat_mode=off` in platformio.ini in order to get the `DallasTemperature` library to load. here is a very simple example that works with `lib_compat_mode = off` in `platformio.ini` but doesn't without it (`error: MyLib.h: No such file or directory`). https://github.com/dolfandringa/pio_arduinofake_unittest

minayaserrano, 2020-02-23:
Same here. The arduino-mock example (https://github.com/platformio/platformio-examples/tree/develop/unit-testing/arduino-mock) is broken. It works adding `lib_compat_mode=off` in platformio.ini.

But this workaround doesn't work to me, because other external dependencies don't work with this configuration. For example, in my project I am using U8g2lib (https://github.com/olikraus/u8g2) and when U8g2lib try to load .h files it seems that it can't find them:
```
In file included from .platformio/lib/U8g2_ID942/src/U8g2lib.cpp:40:0:
.platformio/lib/U8g2_ID942/src/U8g2lib.h:49:10: fatal error: Print.h: No such file or directory
```
I don't know if I'm missing something or if I can help with something.

aviborg, 2021-01-24:
Not sure if it applies to to this issue but seems like I have a similar problem. Using 0.2.2. 

```
Processing test_desktop in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Building...
In file included from .pio\libdeps\native\ArduinoFake\src/arduino/WString.h:25,
                 from .pio\libdeps\native\ArduinoFake\src/arduino/Arduino.h:223,
                 from .pio\libdeps\native\ArduinoFake\src/ArduinoFake.h:13,
                 from .pio\libdeps\native\ArduinoFake\src/Arduino.h:1,
                 from src/HanReader/DlmsReader.h:4,
                 from test\test_desktop\test_main.cpp:1:
.pio\libdeps\native\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6172_F1::enable_if<ArduinoJson6172_F1::is_pointer<T>::value, T>::type ArduinoJson6172_F1::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6172_F1::enable_if<ArduinoJson6172_F1::is_pointer<T>::value, T>::type = const __FlashStringHelper*]':
.pio\libdeps\native\ArduinoJson\src/ArduinoJson/Deserialization/DeserializationError.hpp:103:12:   required from here
.pio\libdeps\native\ArduinoFake\src/arduino/pgmspace.h:106:29: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                            ~^~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\native\test\test_desktop\test_main.o] Error 1
```

askreet, 2021-05-01:
I'm having the same issue @aviborg. This is all a bit lower level than my comfort zone, but ArduinoFake defines `pgm_read_ptr` as:

```c
#define pgm_read_ptr(addr) (*(const void *)(addr))
```

Which seems to ultimately evaluate to a `void` type (versus a `void*`), where casting doesn't make sense (as highlighted by ArduinoJson's error). I wonder if that should really be:

```c
#define pgm_read_ptr(addr) (const void *)(addr)
```

tomec-martin, 2021-08-08:
It seems that it is enougth to change: 
`#define pgm_read_ptr(addr) (*(const void * const *)(addr))`
Pull request is above...

closed 2022-04-10 by FabioBatSilva