FabioBatSilva/ArduinoFake #7: pgmspace.h and noniso.h included in WString.cpp instead of WString.h

dolfandringa, 2020-02-18:
During the struggle with issue #6 I ran into the following. `PROGMEM` is defined in `pgmspace.h` but that file is only included in WString.cpp. Because of it, I ran into an issue with the OneWire library, which uses PROGMEM, and threw these errors:

```
*** [.pio/build/native/lib26a/therma_node/therma_node.o] Error 1
.pio/libdeps/native/OneWire_ID1/OneWire.cpp:496:30: error: expected initializer before ‘dscrc2x16_table’
  496 | static const uint8_t PROGMEM dscrc2x16_table[] = {
      |                              ^~~~~~~~~~~~~~~
.pio/libdeps/native/OneWire_ID1/OneWire.cpp: In static member function ‘static uint8_t OneWire::crc8(const uint8_t*, uint8_t)’:
.pio/libdeps/native/OneWire_ID1/OneWire.cpp:511:23: error: ‘dscrc2x16_table’ was not declared in this scope
  511 |   crc = pgm_read_byte(dscrc2x16_table + (crc & 0x0f)) ^
      |                       ^~~~~~~~~~~~~~~
.pio/libdeps/native/OneWire_ID1/OneWire.cpp:511:9: error: ‘pgm_read_byte’ was not declared in this scope; did you mean ‘read_bytes’?
  511 |   crc = pgm_read_byte(dscrc2x16_table + (crc & 0x0f)) ^
      |         ^~~~~~~~~~~~~
      |         read_bytes
*** [.pio/build/native/lib3ac/OneWire_ID1/OneWire.o] Error 1
```

The solution for me was to `#include pgmspace.h` in `WString.h` instead. I did the same for `noniso.h`.

FabioBatSilva, 2020-02-18:
I think that makes sense.

Are you willing to send a PR with the changes ?

dolfandringa, 2020-02-19:
Sure. Will do this morning

dolfandringa, 2020-02-19:
OK, it became the afternoon, but it's here

FabioBatSilva, 2020-02-19:
Awesome !! 

Thanks @dolfandringa  👍

FabioBatSilva, 2020-02-19:
Closed by https://github.com/FabioBatSilva/ArduinoFake/pull/11

closed 2020-02-19 by FabioBatSilva