FabioBatSilva/ArduinoFake #8: undefined reference to `yield'

dolfandringa, 2020-02-18:
I see yield is defined in `arduino/Arduino.h` but not implemented anywhere. I am using ArduinoFake with DallasTemperature, which uses the `yield` function, which causes the following compile error: `undefined reference to `yield'`. How is yield supposed to be implemented? I don't care what it does (it won't even be called), but my efforts at implementing it  myself with am empty function didn't work.

FabioBatSilva, 2020-02-18:
Hi..

The `yield`  is not supported by `ArduinoFake` yet..

If you want to contribute and implement the missing functionality
You can follow this instructions here on [Contribution Guidelines](https://github.com/FabioBatSilva/ArduinoFake/blob/master/CONTRIBUTING.md)

dolfandringa, 2020-02-19:
Made a PR

FabioBatSilva, 2020-02-19:
Closed by https://github.com/FabioBatSilva/ArduinoFake/pull/10

closed 2020-02-19 by FabioBatSilva