FabioBatSilva/ArduinoFake #12: Is there a way to automatically implement stub all functions?

dolfandringa, 2020-02-19:
External libraries that I am using, are calling a lot of built-in arduino functions (yield/cli/sei/pinMode/etc). I don't care about these calls, but they are resulting in `fakeit::UnexpectedMethodCallException` anyway. Is there a way to automatically just `AlwaysReturn` them or something? Right now I am just debugging it by constantly running the tests, checkig the backtrace of where the tests fail, and add an `AlwaysReturn` in my `int main` for them. This is kind of tedious.

FabioBatSilva, 2020-02-19:
Unless you know what the function should return you are going to get all sorts of random behaviours doing that..
In that case i think you should be mocking the lib if they are compatible with [FakeIt](https://github.com/eranpeer/FakeIt)
or create a a wrapper around the library and mock that.

See : https://github.com/FabioBatSilva/ArduinoFake/blob/master/examples/mock-injection/test/test_my_service.cpp

dolfandringa, 2020-02-20:
Thanks. Yeah, I just mocked a much higher level in the mean time, preventing any of the underlying code being called. I guess I'm still getting used to doing this stuff with Fakeit/C/C++ (coming from much higher level languages)

closed 2020-02-20 by dolfandringa