FabioBatSilva/ArduinoFake #13: Adding SPI Arduino Core Library

rcalcover, 2020-03-02:
Hi I am trying to add [SPI Arduino Core library ](https://github.com/arduino/ArduinoCore-avr/tree/master/libraries/SPI) and contribute in ArduinoFake for the community. 
I have followed the contribution guidelines and added SPI the same way as the `Print` since its quite similar to it. The issue was that I encountered an error that I was stuck on `ArduinoFake.h`:
```In file included from src/Arduino.h:1,
                 from test/main.cpp:1:
src/ArduinoFake.h: In member function ‘SPIClassFake* ArduinoFakeContext::SPIClass(SPIClass*)’:
src/ArduinoFake.h:59:52: error: cannot dynamic_cast ‘instance’ (of type ‘class SPIClass*’) to type ‘class SPIClassFakeProxy*’ (source type is not polymorphic)
         if (dynamic_cast<name##FakeProxy*>(instance)) { \
                                                    ^
src/ArduinoFake.h:103:9: note: in expansion of macro ‘_ArduinoFakeInstanceGetter2’
         _ArduinoFakeInstanceGetter2(SPIClass, SPIClass)
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/ArduinoFake.h:60:59: error: cannot dynamic_cast ‘instance’ (of type ‘class SPIClass*’) to type ‘class SPIClassFakeProxy*’ (source type is not polymorphic)
             return dynamic_cast<name##FakeProxy*>(instance)->get##name##Fake(); \
                                                           ^
src/ArduinoFake.h:103:9: note: in expansion of macro ‘_ArduinoFakeInstanceGetter2’
         _ArduinoFakeInstanceGetter2(SPIClass, SPIClass)` 
```
It showed when I added `_ArduinoFakeInstanceGetter2(SPIClass, SPIClass)` on `ArduinoFakeContext`.

Maybe because SPI Class has static members? But I have removed the `static` modifier on the members of SPIClass but it did not fixed the problem.

I have forked your repo to and made a branch to work on here is the link of my [PR]https://github.com/rcalcover/ArduinoFake/pull/1).

FabioBatSilva, 2020-03-04:
Hi @rcalcover.
Thanks for looking into it..

I think you don't need `_ArduinoFakeInstanceGetter2(SPIClass, SPIClass)` in this case..

Also think that the implementation for `SPIClass::begin` and other static methods need to change.. 
Since you don't have access to `this` you should be using `ArduinoFakeInstance(SPIClass)->begin()` instead of `ArduinoFakeInstance(SPIClass, this)->begin()`

david284, 2022-03-09:
Hi Guys,
Has support for SPI been added to ArduinoFake? - the trail above suggested it was being looked at

Trying to setup testing for a project that includes an SPI CAN bus adapter (MCP2515), and the library for this (ACAN2515) obviously includes SPI.h, which currently throws an error
I'm only testing my code, so currently working around it by removing the include to the library & adding my own define for the data structure that's then missing

nerdyscout, 2023-01-01:
tried to fix this today, but seems to be above my knowledge
https://github.com/nerdyscout/ArduinoFake/tree/feature/spi

r89m, 2023-01-21:
I think you were missing the `==` operator in `SPISettings` - I've implemented it below and the tests now pass for me even with the commented out line being run.
```c++
class SPISettings {
 private:
  uint32_t clock;
  uint8_t bitOrder;
  uint8_t dataMode;
 public:
  SPISettings(uint32_t clock, uint8_t bitOrder, uint8_t dataMode) : clock(clock), bitOrder(bitOrder), dataMode(dataMode) {}
  SPISettings() { SPISettings(4000000, MSBFIRST, SPI_MODE0); }
  friend class SPIClass;

  bool operator==(const SPISettings &other) const {
    return (clock == other.clock) && (bitOrder == other.bitOrder) && (dataMode == other.dataMode);
  }
};
```

FabioBatSilva, 2023-01-22:
Thanks everyone..

PR https://github.com/FabioBatSilva/ArduinoFake/pull/37 got merged, will publish a new version next.