FabioBatSilva/ArduinoFake #14: Struggle creating Serial1 mock

TomasAllegrini, 2020-05-21:
Hi, i've trying to create another serial port with the framework, but could't make it work, i need to mock Serial1.
Is there any tutorial on how to make this mocks for this framework?
Thanks

FabioBatSilva, 2020-05-21:
That is not supported yet..

If you are interested in contributing check out :
https://github.com/FabioBatSilva/ArduinoFake/blob/master/CONTRIBUTING.md

I think that is the same as the [Serial](https://github.com/FabioBatSilva/ArduinoFake/blob/master/src/SerialFake.cpp#L91) implementation just a different variable..

TomasAllegrini, 2020-05-22:
okey, i am still having trouble understanding how to make these mocks, if i can make another Serial, i will create a branch and upload it here.
Any help is welcome 
What i cant' understand, is the relationship with SerialFake, Serial and _Serial, _Serial seems to be the implementation that Serial have to call when invoked, but _Serial and SerialFake should't be the same class?

FabioBatSilva, 2020-05-22:
The `_Serial` header comes from arduino.. 
`SerialFake` is a mockable version that is compatible with [FakeIt](https://github.com/eranpeer/FakeIt) 

There is no concrete implementation for `SerialFake` only mocks.. 
and the implementation of `_Serial` provided by `ArduinoFake` acts like a proxy that just invokes the mock.

closed 2022-04-10 by FabioBatSilva