FabioBatSilva/ArduinoFake #16: I tried it with PlatformIO and get undefined reference to `WinMain@16'

ats3788, 2020-07-27:
Where has the main.cpp File to be
I put it in the src folder.

ats3788, 2020-07-27:
It does not work in my case with the Arduino syntax setup() & loop() you have to change it to main() function.

trullock, 2021-08-29:
I have this exact issue, what is the solution @ats3788 

Thanks

maxgerhardt, 2022-02-05:
When compiling for `platform = native` you need to define the `main()` function to generate a proper executable. The entry point of a normal PC program is not `setup()`. See [official example](https://github.com/platformio/platformio-examples/blob/develop/unit-testing/calculator/test/test_common/test_calculator.cpp#L55-L80) for a way to automatically create the correct entry point.

prof7bit, 2022-04-08:
The examples here in this repository all have setup() and loop() and NO main(), so are these examples wrong?

maxgerhardt, 2022-04-08:
Huh? 

Both of the examples here have the main function. It's intended for testing (`pio test`, sources in `test/`), not compilation of the `src/` program on the native platform. If you want that, you need a `main()`. 

https://github.com/FabioBatSilva/ArduinoFake/blob/910fc5f7dbbdc9d9094aadd29e108bcea063c540/examples/wiring-blink/test/test_main.cpp#L32-L42

https://github.com/FabioBatSilva/ArduinoFake/blob/910fc5f7dbbdc9d9094aadd29e108bcea063c540/examples/mock-injection/test/test_my_service.cpp#L40-L47

So that's alright.

prof7bit, 2022-04-08:
https://github.com/FabioBatSilva/ArduinoFake/blob/master/examples/wiring-blink/src/main.cpp

maxgerhardt, 2022-04-08:
You're pointing at a source in `src/`. This library is intended for testing, the sources in `test/` matter. `src/` is *supposed to be* your embedded firmare, `test/` the tests for your native machine.

FabioBatSilva, 2022-04-10:
Thanks for the clarification @maxgerhardt 

@prof7bit , @ats3788 let me know if it still not clear..

closed 2022-04-10 by FabioBatSilva