FabioBatSilva/ArduinoFake #25: Delay is causing termination

tickietackie, 2021-12-05:
Hello,
I am using the delay function which is causing the following crash when used within my tests.

```
Processing NativeTestWebserver in native environment
------------------------------------------
Building...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ off
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
...
Testing...
libc++abi: terminating with uncaught exception of type fakeit::UnexpectedMethodCallException
```

I tried if it will even crash for a very simple test, which it does for me:
```
#define NATIVE_TEST
#include <ArduinoFake.h>
#include <unity.h>

void test_wifi_password(void) {
  int test = 0;
  TEST_ASSERT_EQUAL(test, 0);
}

int main(int argc, char** argv) {
  UNITY_BEGIN();
  delay(2000);

  RUN_TEST(test_wifi_password);

  UNITY_END();

  return 0;
}
```

When I remove the delay(2000) it runs smoothly. Is this a general error, or can somebody explain the behavior?

I am running this on a Mac M1 and with the current master branch on GitHub.

wrong-kendall, 2021-12-06:
You need to tell the mock to expect the call and what to do when the method is called.

```
void test_something_with_a_delay(){
  When(Method(ArduinoFake(), delay)).AlwaysReturn();
  delay(1000);
  Verify(Method(ArduinoFake(), delay).Using(1000)).Exactly(1);
}
```

FabioBatSilva, 2022-04-10:
Thanks for the explanation @wrong-kendall ..

@tickietackie as mentioned.. you have to mock the calls

closed 2022-04-10 by FabioBatSilva