FabioBatSilva/ArduinoFake #31: Serial.begin() and Serial.print() are crashing

prof7bit, 2022-04-08:
````C++
#include <ArduinoFake.h>

int main() {
    Serial.begin(9600);
    Serial.print("hello");
}
````

````
$ .pio/build/native/program 
Segmentation fault (core dumped)
````

I would have expected Serial.begin() to be a no-op, or maybe just set a flag that begin was called, and Serial.print() to simply write to stdout. Is this an unreasonable assumption?

FabioBatSilva, 2022-04-10:
No, you have to mock the functions you are calling..

The intention of this lib is to provide a way to mock the arduino and run tests outside of an arduino microcontroller..

maybe some work needs to be done on the docs to make that more clear. 

but this is not a actual implementation, you wont be able o run the native program..
only via `pio test`

closed 2022-04-10 by FabioBatSilva

MacDada, 2022-12-24:
> maybe some work needs to be done on the docs to make that more clear.

yea… so… how do I do it?

I want to test a function that uses `Serial.print()`. I have `#include <ArduinoFake.h>` at the beginning of the test file. What's next?

FabioBatSilva, 2022-12-24:
Mock the calls..
Will be something similar to this one.. but using `Serial`

https://github.com/FabioBatSilva/ArduinoFake/blob/master/examples/mock-injection/test/test_my_service.cpp#L18