FabioBatSilva/ArduinoFake #39: Unit testing Wire.h with UNO build

david284, 2023-02-09:
I’m creating unit tests in PlatformIO using Unity with the ArduinoFake library I’m trying to test a piece of code that uses the Wire library
I have built a simple function in a separate file that just calls Wire.begin(), and has _#include <Wire.h>_, which builds fine for UNO
When I try to run the unit test (env:native), it complains that the build cannot find <Wire.h>
If I comment out _#include <Wire.h>_ in the code under test, then it builds native & the test passes ok - but now the UNO build fails

It looks like I'm using this wrong - how should I be doing this?

my platformio.ini file has the following…

```
[env:native]
lib_deps = fabiobatsilva/ArduinoFake@^0.3.1

```
My code under test

```
#include <Arduino.h>
#include <Wire.h>

#include "protoWire.h"

void startWire(void) {
    Wire.begin();
};

```
My testing code

```
#include <Arduino.h>
#include <unity.h>
using namespace fakeit;

#include "protoWire.h"

void setUp (void) {} /* Is run before every test, put unit init calls here. */
void tearDown (void) {} /* Is run after every test, put unit clean-up calls here. */

void test_startWire(void){
    ArduinoFakeReset();
    When(OverloadedMethod(ArduinoFake(Wire), begin, void(void))).AlwaysReturn();
    startWire();
    Verify(OverloadedMethod(ArduinoFake(Wire), begin, void())).Once();
}

int main(int argc, char **argv)
{
   UNITY_BEGIN();
    printf("***** starting WIRE tests *****\n\n");

    RUN_TEST(startWire);

    return UNITY_END();
}

```

david284, 2023-02-09:
Ok - found a solution - wrapped the include in an #ifndef 
Not pretty, but does the job

```
#ifndef PIO_UNIT_TESTING
    #include <Wire.h>
#endif
```

wrong-kendall, 2023-03-20:
Interestingly, I think it may be related to PlatformIO dependency management and not necessarily ArduinoFake (though it may be possible to adjust something on the ArduinoFake side). It works for me if I clone ArduinoFake directly into my platformio's `lib/` but not when I let platformio find and install the dependency. I also tried switching from the default `ldf_mode` of `chain` to `deep`.
```
[15:55:09 03/20/23]: <~/development/fake_includes/lib>$ git clone git@github.com:wrong-kendall/ArduinoFake.git
Cloning into 'ArduinoFake'...
remote: Enumerating objects: 420, done.
remote: Counting objects: 100% (145/145), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 420 (delta 108), reused 109 (delta 94), pack-reused 275
Receiving objects: 100% (420/420), 140.71 KiB | 355.00 KiB/s, done.
Resolving deltas: 100% (215/215), done.
[15:55:33 03/20/23]: <~/development/fake_includes>$ pio test -e native 
Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing * in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
test/test_native.cpp:21: test_startWire [PASSED]
------------------------------------------------------------------------ native:* [PASSED] Took 4.13 seconds ------------------------------------------------------------------------

====================================================================================== SUMMARY ======================================================================================
Environment    Test    Status    Duration
-------------  ------  --------  ------------
native         *       PASSED    00:00:04.135
===================================================================== 1 test cases: 1 succeeded in 00:00:04.135 =====================================================================
```

protoWire.h:
```
#pragma once

void startWire();
```

protoWire.cpp:
```
#include <Arduino.h>
#include <Wire.h>

#include "protoWire.h"

void startWire(void) { Wire.begin(); };
```
test_native.cpp:
```
#include "protoWire.h"
#include <Arduino.h>
#include <unity.h>
using namespace fakeit;

void setUp(void) {} /* Is run before every test, put unit init calls here. */
void tearDown(void) {
} /* Is run after every test, put unit clean-up calls here. */

void test_startWire(void) {
  ArduinoFakeReset();
  When(OverloadedMethod(ArduinoFake(Wire), begin, void(void))).AlwaysReturn();
  startWire();
  Verify(OverloadedMethod(ArduinoFake(Wire), begin, void())).Once();
}

int main(int argc, char **argv) {
  UNITY_BEGIN();
  printf("***** starting WIRE tests *****\n\n");

  RUN_TEST(test_startWire);

  return UNITY_END();
}
```
platform.ini:
```
[env:adafruit_feather_m4]
platform = atmelsam
board = adafruit_feather_m4
framework = arduino

[env:native]
platform = native
;lib_deps = fabiobatsilva/ArduinoFake@^0.3.1
```

Then, removing all of the deps and the local version of ArduinoFake, the code fails:
```
[15:58:20 03/20/23]: <~/development/fake_includes>$ rm -rf lib/ArduinoFake 
[16:03:45 03/20/23]: <~/development/fake_includes>$ rm -rf .pio/
[16:03:57 03/20/23]: <~/development/fake_includes>$ pio test -e native
Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing * in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Library Manager: Installing throwtheswitch/Unity @ ^2.5.2
Unpacking  [####################################]  100%
Library Manager: Unity@2.5.2 has been installed!
lib/protoWire/protoWire.cpp:2:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************

    2 | #include <Wire.h>
      |          ^~~~~~~~
compilation terminated.
*** [.pio/build/native/lib91a/protoWire/protoWire.o] Error 1
Building stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
----------------------------------------------------------------------- native:* [ERRORED] Took 3.02 seconds -----------------------------------------------------------------------

====================================================================================== SUMMARY ======================================================================================
Environment    Test    Status    Duration
-------------  ------  --------  ------------
native         *       ERRORED   00:00:03.022
===================================================================== 1 test cases: 0 succeeded in 00:00:03.022 =====================================================================
```

platform.ini:
```
[env:adafruit_feather_m4]
platform = atmelsam
board = adafruit_feather_m4
framework = arduino

[env:native]
platform = native
lib_deps = fabiobatsilva/ArduinoFake@^0.3.1
```