FabioBatSilva/ArduinoFake #40: Any intention to support an EEPROM 'fake'

david284, 2023-02-14:
One of the libraries I'm using expects 'EEPROM.h'
Is there any intention to support a version of this in ArduinoFake?

FabioBatSilva, 2023-02-16:
Hi,

I'm not planing on working on it, 
But I think it should be supported.

But there are instructions here explaining how to make a PR adding support :
https://github.com/FabioBatSilva/ArduinoFake/blob/master/CONTRIBUTING.md

It will looks something like this:
https://github.com/FabioBatSilva/ArduinoFake/pull/26/files

wraybowling, 2023-03-07:
Wow what a coincidence! I just dug up a project I abandoned 3 years ago and now remembering it was because I did not understand how to write a test for a file that uses `<EEPROM.h>`. I thought surely there's a way and have burned up a couple horus here still not finding a solution.

@FabioBatSilva when you say "i think it should be supported" do you mean that you think it already should work, or are you suggesting that we should try to add it ourselves?

FabioBatSilva, 2023-03-07:
@wraybowling I mean something we should support. 

Yes, something that you or someone else interested can contribute following the example/doc i linked.

wrong-kendall, 2023-05-17:
I started playing with making ArduinoFake [support EEPROM](https://github.com/wrong-kendall/ArduinoFake/commit/16f286a954a32114a2682876c5366c69dbbee63c) but quickly realized that the biggest problem is going to be the use of templates. Since you can't use templates and virtual methods, there's really no way to support the ability to read and write arbitrary datatypes.

@david284 How does the library you're using interact with `EEPROM.h`?

@FabioBatSilva If you think it's worth it to have a partially supported EEPROM, I can send a PR for my limited functionality one. It would at least be a start in case someone wanted to add more functionality (like array access, which I didn't implement) but I kinda hate adding something that's "broken"

david284, 2023-05-18:
The library is using byte orientated read/writes I believe

   byte readEEPROM(unsignedinteeaddress);
voidwriteEEPROM(unsignedinteeaddress, byte data);
   byte readBytesEEPROM(unsignedinteeaddress, byte nbytes, byte dest[]);
voidwriteBytesEEPROM(unsignedinteeaddress, byte src[], byte numbytes);

If that's what you're asking

Regards
David

wrong-kendall, 2023-05-19:
Thanks, that's exactly what I was asking and that's the only functionality that my change implements.