FabioBatSilva/ArduinoFake #43: How to build this project?

MacDada, 2023-04-30:
I'm still noobish with C++ – I mostly use PlatformIO, never wrote a single line of Makefile.

So, the question is: how do I build _this_ project?

I've seen [CONTRIBUTING.md](https://github.com/FabioBatSilva/ArduinoFake/blob/master/CONTRIBUTING.md), but it's inconclusive.

I've tried using `make` (with no success):

```zsh
➜  ArduinoFake git:(master) ✗ make
make: cmake: No such file or directory
make: *** [cmake] Error 1
➜  ArduinoFake git:(master) ✗ mkdir build
➜  ArduinoFake git:(master) ✗ make       
make: cmake: No such file or directory
make: *** [cmake] Error 1
➜  ArduinoFake git:(master) ✗ make all
make: cmake: No such file or directory
make: *** [cmake] Error 1
➜  ArduinoFake git:(master) ✗ make build
make: cmake: No such file or directory
make: *** [cmake] Error 1
```

I've tried using `pio run` (kinda works, but there is no `main()`):

```zsh
➜  ArduinoFake git:(master) ✗ pio run
Processing native (platform: native)
-------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/native/src/ArduinoFake.o
Compiling .pio/build/native/src/ClientFake.o
Compiling .pio/build/native/src/FunctionFake.o
Compiling .pio/build/native/src/PrintFake.o
Compiling .pio/build/native/src/SPIFake.o
Compiling .pio/build/native/src/SerialFake.o
Compiling .pio/build/native/src/StreamFake.o
Compiling .pio/build/native/src/WireFake.o
Compiling .pio/build/native/src/arduino/IPAddress.o
Compiling .pio/build/native/src/arduino/WString.o
Compiling .pio/build/native/src/arduino/noniso.o
Linking .pio/build/native/program
Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
*** [.pio/build/native/program] Error 1
===================================== [FAILED] Took 6.83 seconds =====================================

```

I've tried using `pio test` (finally something working as expected):

```zsh
➜  ArduinoFake git:(master) pio test
Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing * in native environment
-------------------------------------------------------------------------------------------------------
Building...
Library Manager: Installing throwtheswitch/Unity @ ^2.5.2
Unpacking  [####################################]  100%
Library Manager: Unity@2.5.2 has been installed!
Testing...
test/main.cpp:33: ArduinoStringTest::test_constructors  [PASSED]
test/main.cpp:127: ArduinoContextTest::test_single_instance     [PASSED]
test/main.cpp:128: ArduinoContextTest::test_reset       [PASSED]
test/main.cpp:129: ArduinoContextTest::test_function_mock       [PASSED]
test/main.cpp:130: ArduinoContextTest::test_print_mock  [PASSED]
test/main.cpp:131: ArduinoContextTest::test_stream_mock [PASSED]
test/main.cpp:132: ArduinoContextTest::test_serial_mock [PASSED]
test/main.cpp:133: ArduinoContextTest::test_getter_overload_with_proxy  [PASSED]
test/main.cpp:134: ArduinoContextTest::test_getter_overload_with_mapping        [PASSED]
test/main.cpp:135: ArduinoContextTest::test_unknown_instance_exception  [PASSED]
test/main.cpp:187: FunctionTest::test_timestamps        [PASSED]
test/main.cpp:188: FunctionTest::test_pin_mode  [PASSED]
test/main.cpp:189: FunctionTest::test_digital_pin       [PASSED]
test/main.cpp:190: FunctionTest::test_analog_pin        [PASSED]
test/main.cpp:191: FunctionTest::test_delay     [PASSED]
test/main.cpp:192: FunctionTest::test_detach    [PASSED]
test/main.cpp:193: FunctionTest::test_attach    [PASSED]
test/main.cpp:194: FunctionTest::test_cli       [PASSED]
test/main.cpp:195: FunctionTest::test_sei       [PASSED]
test/main.cpp:196: FunctionTest::test_pulsein   [PASSED]
test/main.cpp:197: FunctionTest::test_shift     [PASSED]
test/main.cpp:198: FunctionTest::test_random    [PASSED]
test/main.cpp:199: FunctionTest::test_tone      [PASSED]
test/main.cpp:200: FunctionTest::test_map       [PASSED]
test/main.cpp:100: PrintTest::test_print_variables      [PASSED]
test/main.cpp:101: PrintTest::test_println_variables    [PASSED]
test/main.cpp:168: StreamTest::test_extends_print       [PASSED]
test/main.cpp:169: StreamTest::test_stream_basics       [PASSED]
test/main.cpp:170: StreamTest::test_stream_find [PASSED]
test/main.cpp:171: StreamTest::test_stream_parse        [PASSED]
test/main.cpp:172: StreamTest::test_stream_read [PASSED]
test/main.cpp:80: SerialTest::test_extends_stream       [PASSED]
test/main.cpp:81: SerialTest::test_global_serial        [PASSED]
test/main.cpp:82: SerialTest::test_basics       [PASSED]
test/main.cpp:103: WireTest::test_extends_stream        [PASSED]
test/main.cpp:104: WireTest::test_global_wire   [PASSED]
test/main.cpp:105: WireTest::test_basics        [PASSED]
test/main.cpp:35: SpiTest::test_basics  [PASSED]
test/main.cpp:141: ClientTest::test_basics      [PASSED]
test/main.cpp:142: ClientTest::test_connect     [PASSED]
test/main.cpp:143: ClientTest::test_write       [PASSED]
test/main.cpp:144: ClientTest::test_read        [PASSED]
test/main.cpp:145: ClientTest::test_inject_instance     [PASSED]
test/main.cpp:13: IncludeTest::test_empty       [PASSED]
-------------------------------- native:* [PASSED] Took 61.63 seconds --------------------------------

=============================================== SUMMARY ===============================================
Environment    Test    Status    Duration
-------------  ------  --------  ------------
native         *       PASSED    00:01:01.629
============================= 44 test cases: 44 succeeded in 00:01:01.629 =============================
```

Anything else? Should I just run `pio test`, or do I somehow make `make` working?

I'm on MacOS with CLion, if that changes anything…

Teagan42, 2023-05-17:
`make` requires `cmake` - install that via apt or brew or yum (whatever package manager your system uses)

RobertByrnes, 2023-05-17:
When I did this I found doing it in Windows was a pain due to 'make' in that environment, however, cloning the repo into WSL and installing make and cmake there worked ok!

`cmake .` and then `make` to build, followed by make test to run the test suites

MacDada, 2023-12-08:
> `make` requires `cmake` - install that via apt or brew or yum (whatever package manager your system uses)

yeah, i had to install it via `brew install cmake` and now `make` works as expected, thx!

closed 2023-12-08 by MacDada