FabioBatSilva/ArduinoFake #53: Running mocks on device with FakeIt

nstansbury, 2023-09-22:
Is it possible to use the FakeIt impl. built by ArduinoFake to run mocks on the device?  Otherwise it seems you need to install and build FakeIt separately to get on device mocks

FabioBatSilva, 2023-09-22:
You trying to run the tests with ArduinoFake on a dev board ?
This is mean to run on native platform (your desktop or ci)

nstansbury, 2023-09-23:
Yes, I want to Spy on some class function calls when running tests on the actual device, and am trying to avoid having to build and configure FakeIt directly, as all that hardwork is already done by `ArduinoFake`.

I think the only solution I have found so far is to add `ArduinoFake` into my `[env:pico]` and then disable `lib-compat-mode` to force PlatformIO to include it when building for the device.