FabioBatSilva/ArduinoFake #56: constructor required for `unordered_map`?

graial, 2023-12-13:
Greetings, 
I tried to `#include "ArduinoFake.h"` in a project tested with cpputest (both from docker & native install) and encountered this error.  

> compiling MyFirstTest.cpp
> In file included from /usr/local/include/c++/12.2.0/bits/hashtable.h:35,
>                  from /usr/local/include/c++/12.2.0/unordered_map:46,
>                  from mocks/ArduinoFake/src/ArduinoFake.h:8,
>                  from tests/MyFirstTest.cpp:3:
> /usr/local/include/c++/12.2.0/bits/hashtable_policy.h: In member function 'void std::__detail::_Local_iterator_base<_Key, _Value, _ExtractKey, _Hash, _RangeHash, _Unused, false>::_M_init(const __hash_code_base&)':
> /usr/local/include/c++/12.2.0/bits/hashtable_policy.h:1495:15: error: expected type-specifier before 'this'
>  1495 |       { ::new(this->_M_h()) __hash_code_base(__base); }
>       |               ^~~~
> compilation terminated due to -Wfatal-errors.
> 

Searching the internet, I've noticed threads like [this](https://stackoverflow.com/questions/27242955/getting-a-expected-type-specifier-error-when-i-instantiate-a-custom-class-in-a) which mention the need for a default constructor.  This is pushing me into areas of C++ I am very unfamiliar with.   
Is this something i can do during test set-up?  If so, how?  

I created minimal example [here](https://github.com/graial/blink-arduino-cpputest)

Many thanks for any feedback or suggestions.

FabioBatSilva, 2023-12-13:
Hi,

Might be something related to cpp 12.

You might wanna start with one of the [examples](https://github.com/FabioBatSilva/ArduinoFake/tree/master/examples) and [-std=gnu++17](https://github.com/FabioBatSilva/ArduinoFake/blob/master/examples/wiring-blink/platformio.ini#L4C1-L4C27)

graial, 2023-12-14:
Good evening and thank you for your response. No luck with this though these are the default compiler flags used by cpputest

-pedantic
-Wno-missing-prototypes
-Wno-strict-prototypes
-Wno-c++14-compat
--std=gnu++17  // was originally --std=c++11
-Wno-c++98-compat-pedantic
-Wno-c++98-compat

basvodde, 2023-12-14:
So... if you comment out all of the CppUTest code and just do the #include of the ArduinoFake then it already occurs ?

graial, 2023-12-14:
yes, correct.
and if that #include is removed, the test suite completes successfully

So there seems to be something in the makefile or compiler chain that is unhappy

graial, 2023-12-18:
The ArduinoFake example projects and also (using Unity) work for me.  

I attempted to use the same approach to build cpputest in [this repo](https://github.com/graial/ArduinoFake-cpputest) and it fails on the same error.

graial, 2023-12-28:
I've now gotten this to a [minimal reproducible example](https://github.com/graial/ArduinoFake-cpputest) via cmake. 

i believe that the lines:
`set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)`

establish the `-std=gnu++17` specification

cheers

graial, 2024-03-15:
Greetings @FabioBatSilva, just wanted to mention that with some help on the cpputest forum, I was able to clear this.

https://github.com/graial/ArduinoFake-cpputest

my forked repo is now building but encountering a different error which I haven't yet investigated.  If I can't sort that out, I'll open up a different issue when i get the chance

cheers

closed 2024-03-15 by graial