FabioBatSilva/ArduinoFake #61: Unable to start test and view log

ShettyHarapanahalli, 2024-07-29:
I am new to Arduino and trying to mock the application with this library. When I execute the test cases, I get the below message; when I try to check the JSON, it does not exist. Can you please suggest a way forward.     

------------ pcSimulation:test_radiaion [ERRORED] Took 3.07 seconds ------------
=================================== SUMMARY ===================================
Environment    Test           Status    Duration
-------------  -------------  --------  ------------
pcSimulation   test_radiaion  ERRORED   00:00:03.067
================== 1 test cases: 0 succeeded in 00:00:03.067 ==================
 -> Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing test_radiaion in pcSimulation environment
--------------------------------------------------------------------------------
Building...

Saved JSON report to the C:\Users\10364462\.platformio\.cache\tmp\test-list-72429.json

FabioBatSilva, 2024-07-29:
Try increasing the log level.

`pio test -vvv`

closed 2024-07-29 by FabioBatSilva

ShettyHarapanahalli, 2024-07-29:
Yes, I call the test with "pio test -vvv", and the error is the same.