FabioBatSilva/ArduinoFake #62: SerialFake: setDebugOutput() - Methode missing

JoJos1220, 2024-10-09:
Well, the described Method setDebugOutput() is missing within SerialFake....
It actually doesen't matter for functionallity, but for a complete mock of Serial functionallity it should be added.

FabioBatSilva, 2024-10-09:
In case you are interested in contributing  : https://github.com/FabioBatSilva/ArduinoFake/blob/master/CONTRIBUTING.md

JoJos1220, 2024-10-19:
Well, i added it locally for me and it is working fine so far.
Unfortunatly, if i try to make the changes within the original lib, for me it is absolutly not possible to build the lib and execute the testing environment....never compiled projects using cmake, makefiles....

How can i built this thing with cmake? Can i run the unit testing environment in platformio? I always got built errors, or warnings.....

JoJos1220, 2024-11-12:
I opened the Pull Request now with my local changes. Please can somebody test&confirm it for me, because CMake does not work for me... #63